MotorolaMobilityLLC/kernel-msm #600: Kernel Source for Motorola Edge 50 Pro - U2UM34M.27-38-5

karthik1896, 2024-04-14:
### Build ID

Build ID : U2UM34M.27-38-5
Hey! Please release the kernel/vendor/device trees and all the necessary audio/video/display packages for the Motorola Edge 50 Pro.

DhineshCool, 2024-04-14:
We need kernel source release for working on custom rom ,so please release all necessary packages for building

bhargavpydimalla, 2024-04-15:
please release full kernel sources for working on custom rom ,so please release all necessary packages for building

@asbraga-motorola @chenyt9 @awheeler-mmi @mbissaromoto @TinnoOSS

ak-ag, 2024-04-17:
Hey moto please release the kernel and trees for custom rom development

FPSensor, 2024-04-22:
Please release the edge 50 pro kernel so we can study it, remember that it is protected by a GPL v2 license

chenyt9, 2024-05-11:
Hi，@all，
Version here：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UM34.27-38-5

closed 2024-05-11 by chenyt9