MotorolaMobilityLLC/kernel-msm #604: Kernel Source for T1TGNS33.60-114-5 (genevn)

damnthefall, 2024-04-24:
### Build ID

T1TGNS33.60-114-5

chenyt9, 2024-05-17:
@damnthefall ，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1TGNS33.60-114-5

closed 2024-05-17 by chenyt9

damnthefall, 2024-05-17:
Is there any chance we could get the techpacks and everything else required to fully build this release? Along with some build instructions? Thank you in advance

vindicatorr, 2024-09-29:
I'm wondering the same.

It looks like things are moving towards using `Bazel` to build (noted by the `build.config*` files), but it doesn't look possible to run bazel in there without a workspace.

Looking at https://github.com/MotorolaMobilityLLC/kernel-common, I DO see the workspace as `BUILD.bazel`, but there is no tag for this build id there.

You can't simply `source build.config.msm.parrot` to set up the environment because `/common/build.config.common: No such file or directory`. There is no `common` folder, and the config is in the root of the repo.

It sounds like it's WIP, but the wild thing is the last commits for those `build.config*` is in terms of YEARS!