MotorolaMobilityLLC/kernel-msm #613: Kernel Source for Moto Edge 50 Fusion <U2UUI34.40-24-6>

Rakhshan7070, 2024-05-26:
### Build ID

U2UUI34.40-24-6

### Additional

This is for kernel tweaking to get most out of it and for custom rom development projects.

parasy787898, 2024-05-30:
please release full kernel sources for working on custom rom ,so please release all necessary packages for building

Darshil-154, 2024-06-18:
please release full kernel source for custom rom development for moto edge 50 fusion please

DhineshCool, 2024-07-03:
Kindly update with new tag U2UUI34.40-64-1

lingsen92, 2024-07-23:
code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UUI34.40-24-6

lingsen92, 2024-07-24:
version released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UUI34.40-64-1

closed 2024-07-24 by lingsen92

stylemanik, 2024-08-08:
@DhineshCool 
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UUI34.40-64-1

ftoji879-ai, 2026-10-23:
motorola edge 50 fusion versión Latam 16 (W1UUS36H.21-14-4-1-6).  It has the kernel for this phone model because I need it to activate the KernelSu app.

asbraga-motorola, 2026-10-26:
please open a new request for that version