MotorolaMobilityLLC/kernel-msm #616: [HELP] How to merge moto g34 5g (fogos) defconfig?

UlekBR, 2024-06-07:
### Build ID

Of all the attempts I made, one was successful:

```
./scripts/kconfig/merge_config.sh -m \
    arch/arm64/configs/vendor/genericarmv8-64_defconfig \
    arch/arm64/configs/vendor/moto-consolidate.config \
    arch/arm64/configs/vendor/ext_config/moto-holi.config \
    arch/arm64/configs/vendor/ext_config/moto-holi-fogos.config
```

but it didn't start, I received bootloop

I tried other combinations, with other files, but in all the other attempts I made, the kernel does not compile, to be more specific, it gives an error most of the time when compiling the drivers.

I hope someone can help me

SantoshKumarOjha, 2024-06-20:
Are you developing kernel for fogos?

SantoshKumarOjha, 2024-06-21:
https://github.com/AnandSuresh02
He is actively working on this device's kernel. You can speak with him.

closed 2024-07-24 by asbraga-motorola