MotorolaMobilityLLC/kernel-msm #623: Kernel Source for T1SHS33.35-23-20-4-18

LucyfMorning, 2024-06-24:
### Build ID

Build ID: T1SHS33.35-23-20-4-18

### Additional

Hello,

Please release the kernel tree, device tree and all the necessary audio/video/display techpack for Motorola Edge 30 pro (Hiphi) version: T1SHS33.35-23-20-4-18

Model number: XT2201-1
Codename: HIPHI
Android: 13
Version: T1SHS33.35-23-20-4-18 (latest version as of writing)
Variant: RETBR

While processing this request, please take care to include all of the repositories that correspond to ones such as mentioned in https://github.com/MotorolaMobilityLLC/kernel-msm/issues/516 (referenced just as example of what sort of repo's from the tree appear to be easy to miss during a GPL release)

LucyfMorning, 2024-07-01:
Update to latest release T1SHS33.35-23-20-4-20

asbraga-motorola, 2024-07-02:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SHS33.35-23-20-4-18

For T1SHS33.35-23-20-4-20, could you please open a new issue?