MotorolaMobilityLLC/kernel-msm #629: Kernel Source for U3UOC34.49-29-1

AnandSuresh02, 2024-07-14:
### Build ID

U3UOC34.49-29-1

### Additional

Hello,

Please release the kernel tree, device tree and all the necessary audio/video/display techpack for Motorola Moto g85 5G(malmo) version: U3UOC34.49-29-1

Model number: XT2427-4
Codename: malmo
Android: 14
Version: U3UOC34.49-29-1 (latest version as of writing)

While processing this request, please take care to include all of the repositories that correspond to ones such as mentioned in [#516](https://github.com/MotorolaMobilityLLC/kernel-msm/issues/516) (referenced just as example of what sort of repo's from the tree appear to be easy to miss during a GPL release)

lingsen92, 2024-07-24:
version released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U3UOC34.49-29-1

closed 2024-07-24 by lingsen92

tmhakim, 2025-08-16:
Where is the git history for this kernel source?!