MotorolaMobilityLLC/kernel-msm #631: Help for requesting Lenovo mobile kernel source

Mashopy, 2024-07-18:
Hello, I know it has nothing to do with Motorola directly but since Lenovo own Motorola, I ask here since someone may know how I can contact Lenovo directly for my matter.

But does someone know how I can get a copy of the kernel source for the Lenovo Legion Y70 ? Lenovo OpenSource portal doesn't have a link for it at all and doesn't have any steps or guide for requesting them a copy of the code.

asbraga-motorola, 2024-08-09:
Hi @Mashopy ,
We don't have information for Lenovo products, only Motorola.

I did a quick web search and found this: https://support.lenovo.com/br/en/solutions/ht511330-lenovo-open-source-portal
Maybe it has what you are looking for.

Anyway, I'll add comments to this issue if I get more details.

closed 2024-08-09 by asbraga-motorola

Mashopy, 2024-08-10:
> Hi @Mashopy , We don't have information for Lenovo products, only Motorola.
> 
> I did a quick web search and found this: https://support.lenovo.com/br/en/solutions/ht511330-lenovo-open-source-portal Maybe it has what you are looking for.
> 
> Anyway, I'll add comments to this issue if I get more details.

Hi, thank you for your answer but unfortunately it isn't what I'm looking for.
I am aware that Lenovo share their kernel source tarballs there but I seek for a way to contact Lenovo to release a kernel tarball for one of their android device that is only sold on china market which isn't on their open source portal for some weird reason.