MotorolaMobilityLLC/kernel-msm #634: Kernel Source for U1TPNS34.26-78-6-1

Recidl, 2024-08-09:
### Build ID

U1TPNS34.26-78-6-1

### Additional

Please provide the Kernel, Device, and similar trees as well as drivers and vendor that would be used in the development for building LineageOS 21. 

Variant: RETUS
Phone: Motorola Moto G 5G 2023
Carrier: Boost Mobile
Build: U1TPNS34.26-78-6-1 
Android Vers: 14

JuanDavid006, 2024-08-27:
Android 13 version of this same device: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1TPN33.58-68-1-2-1

asbraga-motorola, 2024-08-31:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U1TPNS34.26-78-6-1

closed 2024-08-31 by asbraga-motorola

JuanDavid006, 2024-09-01:
Thank you so much. By the way, any reason why my build is only 24Mb and not 46Mb like the original? I might not be getting the proper defconfig or I'm doing something wrong...