MotorolaMobilityLLC/kernel-msm #637: Fix Long exposure for U3UO34.49-40-1

RahulPalXDA, 2024-08-20:
### Build ID

U3UO34.49-40-1

### Additional

Hello,

Please fix long exposure for,
Device: MALMO
Build Id: U3UO34.49-40-1

This is observed that, It's not working due to it's returning 1/5s is the highest exposure time.

Thanks in advanced for your assistance.

Fraaxius, 2024-09-19:
If you have any problems, you should report them in Motorola support. This place is basically for requesting opensource kernels and others.

asbraga-motorola, 2024-09-23:
> If you have any problems, you should report them in Motorola support. This place is basically for requesting opensource kernels and others.

Yes, that's the goal of these repos. Thanks!

@RahulPalXDA,
You may try to ask in https://forums.lenovo.com/t5/Motorola-Community/ct-p/MotorolaCommunity or directly at https://www.motorola.com

closed 2024-09-23 by asbraga-motorola