MotorolaMobilityLLC/kernel-msm #644: Kernel Source for U2UMS34.27-38-1-2-3-1

Venkat3620, 2024-09-07:
### Build ID

U2UMS34.27-38-1-2-3-1

### Additional

Please publish required dts, techpacks , motorola-modules and others.
codename: eqe
device: motorola edge 50 pro

Venkat3620, 2024-09-12:
Sir, any update on this?
@chenyt9 @awheeler-mmi @mbissaromoto

Venkat3620, 2024-09-18:
@asbraga-motorola

Sourabh-AnanyaTech, 2024-09-19:
I'm following the progress on this issue and was wondering why there's a delay on the source code release. As a potential buyer, I'm really interested in customizing my device and would love to see the sources made available.

Venkat3620, 2024-10-06:
@asbraga-motorola @chenyt9 @awheeler-mmi @mbissaromoto

asbraga-motorola, 2024-10-08:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UMS34.27-38-1-2-3-1

closed 2024-10-08 by asbraga-motorola