MotorolaMobilityLLC/kernel-msm #653: Kernel Source for U1SJ34.2-92-10 (Motorola Edge 30 Fusion / tundra)

ThEMarD, 2024-10-17:
### Build ID

U1SJ34.2-92-10

### Additional

We will be using this for the AOSP known as LineageOS as we already support berlin, berlna, dubai and xpeng and are thinking of adding tundra to that list too

asbraga-motorola, 2024-10-23:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U1SJ34.2-92-10

closed 2024-10-23 by asbraga-motorola

ThEMarD, 2025-11-07:
> I was trying to compile this kernel several times and it always says it's the source tree is not clean. I ran 'make mrproper' several times. It always returns the same error that the source is not clean
> 
> Anyhelp be friendly, I'm a noob.

Please check the readme repo. If there isn't a readme for tundra you can do a new issue there to request it

https://github.com/MotorolaMobilityLLC/readme