MotorolaMobilityLLC/kernel-msm #659: Kernel Source for U2UANS34.72-43-1-3 (codename: avatrn)

Lost-Entrepreneur439, 2024-11-24:
### Build ID

U2UANS34.72-43-1-3

### Additional

This is using kernel 5.10.198, which is very out of date and has nearly 300 CVEs. I love this phone and hate to see it so full of exploits, so I want to try to get it to the latest version of 5.10 (currently 5.10.230)

Alongside the kernel source, it'd also be nice to have the device tree sources, kernel modules, and any other sources that is used by this build, alongside the readme for building too.

Digitalbullits, 2024-11-27:
https://github.com/MotorolaMobilityLLC/motorola-kernel-modules/activity?ref=android-14-release-u2uans34.72-43-1-1-1

Digitalbullits, 2024-11-27:
https://gitgud.io/Van-Firmware-Dumps/motorola/avatrn

asbraga-motorola, 2024-12-03:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UANS34.72-43-1-3

closed 2024-12-03 by asbraga-motorola