MotorolaMobilityLLC/kernel-msm #660: Kernel Source for U3UV34.26-154-5

Kendrenogen, 2024-11-25:
### Build ID

U3UV34.26-154-5

### Additional

As usual, can you release the main kernel source as well as all of the device tree sources, kernel modules, and any other sources that is used by this build?

Also, can you provide a readme as well for kernel compilation?

Finally, can you also release the source of the kernel_platform script and build tools as well like some other OEMs?

liuya9, 2024-11-28:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U3UV34.26-39-8

asbraga-motorola, 2024-12-03:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U3UV34.26-154-5

closed 2024-12-03 by asbraga-motorola

Kendrenogen, 2024-12-03:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U3UV34.26-154-5

Can you also upload a README with compilation instructions?