MotorolaMobilityLLC/kernel-msm #672: Kernel (and device tree) Source for U1UFNS34.41-98-3-11 (fogo)

smaybius, 2025-02-02:
### Build ID

U1UFNS34.41-98-3-11

### Build fingerprint

motorola/fogo_g/fogo:14/U1UFNS34.41-98-3-11/17710-d1a62:user/release-keys

### Additional

I would also like to request a release of the device trees, like both DTS and android_device_motorola_fogo.

asbraga-motorola, 2025-02-20:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U1UFNS34.41-98-3-11

closed 2025-02-20 by asbraga-motorola

smaybius, 2025-02-20:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U1UFNS34.41-98-3-11
@asbraga-motorola 
What about the ones that I listed as well? There could easily be repos that I missed. For the other repos with branches that start with android-14-release-u1ufn34.*, what's different from u1ufns34?

chenyt9, 2025-02-20:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U1UFNS34.41-98-3-11
@smaybius , you can refer to each repo,like:
https://github.com/MotorolaMobilityLLC/kernel-msm-5.4-techpack-camera/releases/tag/MMI-U1UFNS34.41-98-3-11

smaybius, 2025-02-20:
> Version released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U1UFNS34.41-98-3-11 [@smaybius](https://github.com/smaybius) , you can refer to each repo,like: https://github.com/MotorolaMobilityLLC/kernel-msm-5.4-techpack-camera/releases/tag/MMI-U1UFNS34.41-98-3-11

@chenyt9 I only remembered to refer to branches instead of tags, and forgot about the tags even though it was the first way that I've been shown