MotorolaMobilityLLC/kernel-msm #682: Kernel Source for U2UANS34.72-43-1-6 (avatrn)

Lost-Entrepreneur439, 2025-02-21:
### Build ID

U2UANS34.72-43-1-6

### Build fingerprint

motorola/avatrn_g/avatrn:14/U2UANS34.72-43-1-6/b09f7-6b760:user/release-keys

### Additional

Please share any other resources that'd be required to build this kernel.

ethan99757, 2025-04-06:
Please shut it down y'all murdering me I want y'all off my phone 9853609116 and never back on and all the sex traffic y'all needed shut down

asbraga-motorola, 2025-04-09:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UANS34.72-43-1-6

closed 2025-04-09 by asbraga-motorola

Lost-Entrepreneur439, 2025-04-09:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UANS34.72-43-1-6

I asked "Please share the kernel build instructions on the https://github.com/MotorolaMobilityLLC/readme repo.", where is this?