MotorolaMobilityLLC/kernel-msm #685: Kernel Source for V1TR35H.59 (rtwo Android 15 Beta)

BM570, 2025-03-08:
### Build ID

V1TR35H.59

### Build fingerprint

V1TR35H.59

### Additional

It is a closed beta for Android 15 on rtwo, but that doesn't preclude it from GPL release requirement. I want to evaluate and later compare the framework prepping work for Android 15 with the final update that Motorola should be able to globally release soon.

Please release the kernel tree, device tree and all of the audio/video/display techpack for `V1TR35H.59`
Notif: @chenyt9 @awheeler-mmi @mbissaromoto @asbraga-motorola @TinnoOSS

asbraga-motorola, 2025-03-11:
I'm having trouble tracking down any closed beta with that build id.
Can you inform the **ro.product.build.fingerprint** and **ro.mot.build.guid** so I can track down the build?

ah124med, 2025-03-14:
any issus?

ah124med, 2025-03-24:
not coming to my device moto x40

closed 2025-04-03 by asbraga-motorola