MotorolaMobilityLLC/kernel-msm #699: Kernel Source for U2UANS34.72-43-1-7

Lost-Entrepreneur439, 2025-04-13:
### Build ID

U2UANS34.72-43-1-7

### Build fingerprint

motorola/avatrn_g/avatrn:14/U2UANS34.72-43-1-7/f6230-50569:user/release-keys

### Additional

**SHARE THE BUILD INSTRUCTIONS IN https://github.com/MotorolaMobilityLLC/readme**. Seriously. The kernel source code is absolutely useless without this.

Lost-Entrepreneur439, 2025-04-15:
@asbraga-motorola sorry for pinging you, but my last avatrn kernel source request took over 2 months, which is simply unacceptable. by the time i got the kernel source, a new update was already out. i'm hoping pinging you will speed up the process.

once again, please put build instructions in the readme.

asbraga-motorola, 2025-05-06:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UANS34.72-43-1-7

closed 2025-05-06 by asbraga-motorola

Lost-Entrepreneur439, 2025-05-10:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-U2UANS34.72-43-1-7

@asbraga-motorola Once again, my request for the build instructions has been ignored. This is required, the kernel is useless without the instructions. I will ask again, please share the instructions in the readme repo. I will open a new issue if I have to, this behaviour isn't acceptable. I went with Motorola since the bootloader is unlockable, but the refusal to share kernel build instructions kinda defeats the whole reason I purchased this phone.

asbraga-motorola, 2025-05-12:
we are working on the build instructions.