MotorolaMobilityLLC/kernel-msm #701: Kernel Source for V2UX35.47-25

Kendrenogen, 2025-04-18:
### Build ID

V2UX35.47-25

### Build fingerprint

motorola/arcfox_g/arcfox:15/V2UX35.47-25/63266-a28f5:user/release-keys

### Additional

Can you release the source code for the main kernel as well as all of the modules and device trees?

Also, can you release build instructions for the kernel source as well as the modules and device trees?

asbraga-motorola, 2025-05-05:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-V1TR35H.60-33

closed 2025-05-05 by asbraga-motorola

Kendrenogen, 2025-05-06:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-V1TR35H.60-33
@asbraga-motorola That is not the correct source code. The tag is supposed to be V2UX35.47-25

asbraga-motorola, 2025-05-06:
Removed comment due to link to the wrong issue.