MotorolaMobilityLLC/kernel-msm #708: Kernel Source for V2UX35.47-25

Kendrenogen, 2025-05-06:
### Build ID

V2UX35.47-25

### Build fingerprint

motorola/arcfox_g/arcfox:15/V2UX35.47-25/63266-a28f5:user/release-keys

### Additional

https://github.com/MotorolaMobilityLLC/kernel-msm/issues/701 provided the source code of V1TR35H.60-33, not V2UX35.47-25. Can you provide the source code for V2UX35.47-25 specifically.

And as usual the following:

Can you release the source code for the main kernel as well as all of the modules and device trees?

Also, can you release build instructions for the kernel source as well as the modules and device trees?

asbraga-motorola, 2025-05-14:
I'm having trouble tracking down that specific build id.
Can you inform the ro.product.build.fingerprint and ro.mot.build.guid as informed by `adb shell getprop` so I can track down the build?

asbraga-motorola, 2025-05-14:
did you receive this build via OTA? I really cannot find it.
Do you mind sending a file with all properties?

Kendrenogen, 2025-05-14:
@asbraga-motorola 

[propfiles.zip](https://github.com/user-attachments/files/20212695/propfiles.zip)

These are all of the extracted properties

closed 2025-05-21 by asbraga-motorola