MotorolaMobilityLLC/kernel-msm #722: Kernel Source for U2UANS34.72-43-1-8

Lost-Entrepreneur439, 2025-06-25:
### Build ID

U2UANS34.72-43-1-8

### Build fingerprint

motorola/avatrn_g/avatrn:14/U2UANS34.72-43-1-8/ab5eee-04ba4:user/release-keys

### Additional

Can you provide the kernel source for the device as well as the modules and device trees as well?

And can you also provide a README containing compilation instructions for the kernel and all of the modules at https://github.com/MotorolaMobilityLLC/readme?

lunarriaa, 2025-06-27:
Need this too

Lost-Entrepreneur439, 2025-07-14:
@asbraga-motorola thanks for providing the source, but once again, the build instructions are also required under the GPL.

When I asked two months ago, you responded with "we are working on the build instructions", but there’s been no update since. Motorola is obligated under Section 3 of GPLv2 to provide the necessary instructions to make this source buildable (e.g. defconfig, toolchain info, and any custom steps).

This is not optional — it is a legal requirement. Motorola already has a dedicated repository for this purpose (MotorolaMobilityLLC/readme), so I see no reason the instructions can't be posted there.

If the build instructions are not provided within the next 1–2 weeks, I will be forced to escalate this to the Software Freedom Conservancy (SFC) as a formal GPL compliance complaint.

Lost-Entrepreneur439, 2025-07-16:
@chenyt9 Either your instructions or kernel source are broken. The kernel still doesn't build. I have all dependencies that should be needed to build an Android kernel (and i know i do, because ive built android kernels for plenty of devices).

```
ivy@debian:~/kernel_platform/external/dtc$ make all install CC=clang AR=llvm-ar LDFLAGS="--sysroot=$my_top_dir/build/build-tools/sysroot -Wl,-rpath,$my_top_dir/prebuilts/kernel-build-tools/linux-x86/lib64 -L $my_top_dir/prebuilts/kernel-build-tools/linux-x86/lib64 -fuse-ld=lld --rtlib=compiler-rt -fuse-ld=lld --rtlib=compiler-rt" EXTRA_CFLAGS="--sysroot=$my_top_dir/build/build-tools/sysroot -I$my_top_dir/prebuilts/kernel-build-tools/linux-x86/include" PKG_CONFIG=false NO_PYTHON=1 PREFIX=$my_top_dir/out/msm-waipio-parrot-gki/host
         DEP checks.c
         DEP data.c
         DEP dtc.c
         DEP flattree.c
         DEP fstree.c
         DEP livetree.c
         DEP srcpos.c
         DEP treesource.c
         DEP util.c
         LEX dtc-lexer.lex.c
         DEP dtc-lexer.lex.c
         BISON dtc-parser.tab.c
dtc-parser.y: warning: 5 shift/reduce conflicts [-Wconflicts-sr]
dtc-parser.y: note: rerun with option '-Wcounterexamples' to generate conflict counterexamples
         DEP dtc-parser.tab.c
         LEX convert-dtsv0-lexer.lex.c
         DEP convert-dtsv0-lexer.lex.c
         DEP fdtdump.c
         DEP fdtget.c
         DEP fdtput.c
         DEP fdtoverlay.c
         DEP libfdt/fdt.c
         DEP libfdt/fdt_ro.c
         DEP libfdt/fdt_wip.c
         DEP libfdt/fdt_sw.c
         DEP libfdt/fdt_rw.c
         DEP libfdt/fdt_strerror.c
         DEP libfdt/fdt_empty_tree.c
         DEP libfdt/fdt_addresses.c
         DEP libfdt/fdt_overlay.c
         DEP libfdt/fdt_check.c
         DEP libfdt/acpi.c
         DEP tests/get_mem_rsv.c
         DEP tests/root_node.c
         DEP tests/find_property.c
         DEP tests/subnode_offset.c
         DEP tests/path_offset.c
         DEP tests/get_name.c
         DEP tests/getprop.c
         DEP tests/get_prop_offset.c
         DEP tests/get_phandle.c
         DEP tests/get_path.c
         DEP tests/supernode_atdepth_offset.c
         DEP tests/parent_offset.c
         DEP tests/node_offset_by_prop_value.c
         DEP tests/node_offset_by_phandle.c
         DEP tests/node_check_compatible.c
         DEP tests/node_offset_by_compatible.c
         DEP tests/get_alias.c
         DEP tests/get_next_tag_invalid_prop_len.c
         DEP tests/char_literal.c
         DEP tests/sized_cells.c
         DEP tests/notfound.c
         DEP tests/addr_size_cells.c
         DEP tests/addr_size_cells2.c
         DEP tests/appendprop_addrrange.c
         DEP tests/stringlist.c
         DEP tests/setprop_inplace.c
         DEP tests/nop_property.c
         DEP tests/nop_node.c
         DEP tests/sw_tree1.c
         DEP tests/sw_states.c
         DEP tests/move_and_save.c
         DEP tests/mangle-layout.c
         DEP tests/nopulate.c
         DEP tests/open_pack.c
         DEP tests/rw_tree1.c
         DEP tests/rw_oom.c
         DEP tests/set_name.c
         DEP tests/setprop.c
         DEP tests/del_property.c
         DEP tests/del_node.c
         DEP tests/appendprop1.c
         DEP tests/appendprop2.c
         DEP tests/propname_escapes.c
         DEP tests/string_escapes.c
         DEP tests/references.c
         DEP tests/path-references.c
         DEP tests/phandle_format.c
         DEP tests/boot-cpuid.c
         DEP tests/incbin.c
         DEP tests/relref_merge.c
         DEP tests/extra-terminating-null.c
         DEP tests/dtbs_equal_ordered.c
         DEP tests/dtb_reverse.c
         DEP tests/dtbs_equal_unordered.c
         DEP tests/add_subnode_with_nops.c
         DEP tests/path_offset_aliases.c
         DEP tests/utilfdt_test.c
         DEP tests/integer-expressions.c
         DEP tests/property_iterate.c
         DEP tests/subnode_iterate.c
         DEP tests/overlay.c
         DEP tests/overlay_bad_fixup.c
         DEP tests/check_path.c
         DEP tests/check_header.c
         DEP tests/check_full.c
         DEP tests/fs_tree1.c
         DEP tests/truncated_property.c
         DEP tests/truncated_string.c
         DEP tests/truncated_memrsv.c
         DEP tests/asm_tree_dump.c
         DEP tests/value-labels.c
         DEP tests/testutils.c
         DEP tests/trees.S
         DEP tests/dumptrees.c
        CHK version_gen.h
        UPD version_gen.h
         DEP util.c
         DEP dtc-lexer.lex.c
        CHK version_gen.h
         CC srcpos.o
         CC util.o
         CC convert-dtsv0-lexer.lex.o
         LD convert-dtsv0
ld.lld: error: unable to find library -lgcc_s
ld.lld: error: unable to find library -lgcc_s
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:267: convert-dtsv0] Error 1
```
As of right now, Motorola is still violating the GPL and unless this is fixed within 2 weeks, I will still be filing a complaint with the SFC.

lunarriaa, 2025-07-16:
@Lost-Entrepreneur439 I don't know if this would help but I tried building, and when cloning the "kernel-build-tool" the folder is saved as "kernel-build-tool" and not "kernel-build-tools", and also check kernel_platform/prebuilts-master/clang/linux-x86 to check if you have what is specified on line 37 of the readme

Also, line 49 says "msm-waipo-waipo-gki" which seems like an accident.

As for me, I got an error saying "Make: *** No rule to make target 'stdio.h', needed by 'srcpos.o'. Stop." But I'm sure that's a dependency error on my end or something similar.

If it helps, I've had experiences where for some reason Debian will have all dependencies, yet still error, however the same can't be said for Arch, where for me at least it usually always works, so try that? It's worked for me sometimes, and I don't know why.

Lost-Entrepreneur439, 2025-07-16:
> [@Lost-Entrepreneur439](https://github.com/Lost-Entrepreneur439) I don't know if this would help but I tried building, and when cloning the "kernel-build-tool" the folder is saved as "kernel-build-tool" and not "kernel-build-tools", and also check kernel_platform/prebuilts-master/clang/linux-x86 to check if you have what is specified for clang- on line 37 of the readme
> 
> Also, line 49 says "msm-waipo-waipo-gki" which seems like an accident.
> 
> As for me, I got an error saying "Make: *** No rule to make target 'stdio.h', needed by 'srcpos.o'. Stop." But I'm sure that's a dependency error on my end or something similar.
> 
> If it helps, I've had experiences where for some reason Debian will have all dependencies, yet still error, however the same can't be said for Arch, where for me at least it usually always works, so try that? It's worked for me sometimes, and I don't know why.

turns out motorola didn't include the right clang version, grabbing it from here -> https://android.googlesource.com/platform//prebuilts/clang/host/linux-x86/+/b669748458572622ed716407611633c5415da25c/clang-r416183b/ worked.

As for that stdio error, I had that too, Motorola included an extra ../ in a link which causes that error. go to build/build-tools and remove the extra ../ from the sysroot link

lunarriaa, 2025-07-17:
> turns out motorola didn't include the right clang version, grabbing it from here -> [...]

Managed to get past the stdio error, thank you for that, but now I'm getting the following; `make: *** No rule to make target 'gki_defconfig'.  Stop.
`
I'm unsure if I followed the steps right, but on line 51 of the readme we cd into /common/ which has not been made, so I'm unsure if it is supposed to be empty.

chenyt9, 2025-07-17:
Hi，@all，

Thank you for your feedback! We have confirmed and corrected the typographical errors, updating the kernel-build-tools and the path information for msm-waipio-parrot-gki on line 49 to ensure that the compilation can proceed normally.

Regarding the clang tool version issue, thank you very much for providing detailed information and the link. The code library we provide is based on our project's actual situation and uses the Qualcomm open-source library, which is consistent with the information from https://android.googlesource.com/platform//prebuilts/clang/host/linux-x86/+/b669748458572622ed716407611633c5415da25c/clang-r416183b/ . To ensure that you can configure the tool environment correctly, we have updated the correct branch information in the README file.

Please refer to the common code, which is used to build gki, https://github.com/MotorolaMobilityLLC/kernel-common/releases/tag/MMI-U2UANS34.72-43-1-8

Regarding the issue with prebuilts/gas/linux-x86, we understand that while the macro is defined, it is not required for actual use in the project. We have verified that removing the macro allows the compilation to proceed normally. Please ignore this macro and continue with your work.

If you encounter any further issues or need additional assistance with the README or open-source-related matters, please don’t hesitate to contact us. We remain committed to supporting you.

Thank you once again for your feedback and cooperation!

Lost-Entrepreneur439, 2025-07-17:
> [@chenyt9](https://github.com/chenyt9) Unsure if this is on my end, but now when running make all install, I get the following;
> 
> ❯ make all install CC=clang AR=llvm-ar LDFLAGS="--sysroot=$my_top_dir/build/build-tools/sysroot -Wl,-rpath,$my_top_dir/prebuilts/kernel-build-tools/linux-x86/lib64 -L $my_top_dir/prebuilts/kernel-build-tools/linux-x86/lib64 -fuse-ld=lld --rtlib=compiler-rt -fuse-ld=lld --rtlib=compiler-rt" EXTRA_CFLAGS="--sysroot=$my_top_dir/build/build-tools/sysroot -I$my_top_dir/prebuilts/kernel-build-tools/linux-x86/include" PKG_CONFIG=false NO_PYTHON=1 PREFIX=$my_top_dir/out/msm-waipio-parrot-gki/host
> 	  DEP checks.c
> 	  DEP data.c
> 	  DEP dtc.c
> 	  DEP flattree.c
> 	  DEP fstree.c
> 	  DEP livetree.c
> 	  DEP srcpos.c
> 	  DEP treesource.c
> 	  DEP util.c
> 	  LEX dtc-lexer.lex.c
> 	  DEP dtc-lexer.lex.c
> 	  BISON dtc-parser.tab.c
> /run/media/lucine/thing5/android/lineage/kernel_platform/build/build-tools/path/linux-x86/bison: line 7: /run/media/lucine/thing5/android/lineage/kernel_platform/build/build-tools/path/linux-x86/../../../../prebuilts/build-tools/linux-x86/bin/bison: No such file or directory
> 	  LEX convert-dtsv0-lexer.lex.c
> 	  DEP convert-dtsv0-lexer.lex.c
> 	  DEP fdtdump.c
> 	  DEP fdtget.c
> 	  DEP fdtput.c
> 	  DEP fdtoverlay.c
> 	  DEP tests/get_mem_rsv.c
> 	  DEP tests/root_node.c
> 	  DEP tests/find_property.c
> 	  DEP tests/subnode_offset.c
> 	  DEP tests/path_offset.c
> 	  DEP tests/get_name.c
> 	  DEP tests/getprop.c
> 	  DEP tests/get_prop_offset.c
> 	  DEP tests/get_phandle.c
> 	  DEP tests/get_path.c
> 	  DEP tests/supernode_atdepth_offset.c
> 	  DEP tests/parent_offset.c
> 	  DEP tests/node_offset_by_prop_value.c
> 	  DEP tests/node_offset_by_phandle.c
> 	  DEP tests/node_check_compatible.c
> 	  DEP tests/node_offset_by_compatible.c
> 	  DEP tests/get_alias.c
> 	  DEP tests/char_literal.c
> 	  DEP tests/sized_cells.c
> 	  DEP tests/notfound.c
> 	  DEP tests/addr_size_cells.c
> 	  DEP tests/addr_size_cells2.c
> 	  DEP tests/appendprop_addrrange.c
> 	  DEP tests/stringlist.c
> 	  DEP tests/setprop_inplace.c
> 	  DEP tests/nop_property.c
> 	  DEP tests/nop_node.c
> 	  DEP tests/sw_tree1.c
> 	  DEP tests/sw_states.c
> 	  DEP tests/move_and_save.c
> 	  DEP tests/mangle-layout.c
> 	  DEP tests/nopulate.c
> 	  DEP tests/open_pack.c
> 	  DEP tests/rw_tree1.c
> 	  DEP tests/rw_oom.c
> 	  DEP tests/set_name.c
> 	  DEP tests/setprop.c
> 	  DEP tests/del_property.c
> 	  DEP tests/del_node.c
> 	  DEP tests/appendprop1.c
> 	  DEP tests/appendprop2.c
> 	  DEP tests/propname_escapes.c
> 	  DEP tests/string_escapes.c
> 	  DEP tests/references.c
> 	  DEP tests/path-references.c
> 	  DEP tests/phandle_format.c
> 	  DEP tests/boot-cpuid.c
> 	  DEP tests/incbin.c
> 	  DEP tests/extra-terminating-null.c
> 	  DEP tests/dtbs_equal_ordered.c
> 	  DEP tests/dtb_reverse.c
> 	  DEP tests/dtbs_equal_unordered.c
> 	  DEP tests/add_subnode_with_nops.c
> 	  DEP tests/path_offset_aliases.c
> 	  DEP tests/utilfdt_test.c
> 	  DEP tests/integer-expressions.c
> 	  DEP tests/property_iterate.c
> 	  DEP tests/subnode_iterate.c
> 	  DEP tests/overlay.c
> 	  DEP tests/overlay_bad_fixup.c
> 	  DEP tests/check_path.c
> 	  DEP tests/check_header.c
> 	  DEP tests/check_full.c
> 	  DEP tests/fs_tree1.c
> 	  DEP tests/truncated_property.c
> 	  DEP tests/truncated_string.c
> 	  DEP tests/truncated_memrsv.c
> 	  DEP tests/asm_tree_dump.c
> 	  DEP tests/value-labels.c
> 	  DEP tests/testutils.c
> 	  DEP tests/trees.S
> 	  DEP tests/dumptrees.c
> 	  BISON dtc-parser.tab.h
> /run/media/lucine/thing5/android/lineage/kernel_platform/build/build-tools/path/linux-x86/bison: line 7: /run/media/lucine/thing5/android/lineage/kernel_platform/build/build-tools/path/linux-x86/../../../../prebuilts/build-tools/linux-x86/bin/bison: No such file or directory
> 	  CC srcpos.o
> 	  CC util.o
> 	  CC convert-dtsv0-lexer.lex.o
> 	  LD convert-dtsv0
> 	  CC dtc.o
> 	  CC checks.o
> 	  CC data.o
> 	  CC flattree.o
> 	  CC fstree.o
> 	  CC livetree.o
> 	  CC treesource.o
> make: *** No rule to make target 'dtc-parser.tab.h', needed by 'dtc-lexer.lex.o'.  Stop.
> 
> If it helps, I did run line 21 as `git clone --depth=1` instead of the specified `git clone` because previously when getting to this line, the cloning process would take upwards to a hour or 2 and the longest I've ever had being 4hrs and 46 minutes and also resulted in it taking up 40GiB. I only did this for that specific line, as speed for the other `git clone`'s were already fast enough.
> 
> And also, going to build/build-tools/path reveals that there are quite a few broken links at least on my end, said broken links are: zipinfo, xzcat, xz, xxd, xargs, whoami, which, wc, unix2dos, uniq, uname, truncate, true, tr, touch, timeout, test, tee, tail, stat, sort, sleep, sha512sum, sha256sum, sha1sum, setsid, seq, sed, rmdir, rm, realpath, readlink, python3, python2.7, python2, python, pwd, ps, pkill, pgrep, paste, openssl, od, nproc, mv, mktemp, mkdir, md5sum, make, m4, ls, ln, id, hostname, head, gzip, grep, getconf, flex, find, expr, env, egrep, echo, du, dirname, diff, date, cut, cpio, cp, comm, cmp, chmod, cat, bzip2, bzcat, bc, basename, and awk.

yup, same issue here, there's tons of broken links.

lunarriaa, 2025-07-17:
Yeah, I can understand where you're coming from, I had a Google Pixel 6 Pro before the Edge 2024, but it wasn't rootable due to it being Verizon (Even though they say that the device should be unblocked after a few weeks) and so I just had to stick with what I had. However, I switched carriers because Verizon's charges were getting insanely large, and traded the Pixel for the Edge 2024. 

Long story short, I found out it was rootable, rooted it, made recovery images after a while, and now I'm just trying to make custom ROMs for security reasons and also for anyone else who has the phone. 

I got to a point a while ago where I too just stopped caring as the coverage of the device as far as device trees and etc was basically 0 and the troubleshooting was making me go insane. 

Now though, knowing that the goal is somewhat closer has been a little bit of a help, even if support has been kind of iffy, I am still gonna attempt to get this built because I would like the extra security, and also would like to help others who have the phone as well. 

I'll be honest too and say that the support times have somewhat annoyed me, as I usually end up troubleshooting past the person committing to the readme repo's hours, only to end up with a bug and have to wait 24 hours until 3AM EST for something to finally happen. All I wish for from Motorola is if the support was better, and I hope that it does get better soon. 

And knowing that this comment is probably being read by at least one of them, I'd like to ask if they could at least try building on a personal system or a test device with a OS like Debian to see if they encounter any issues before they comment saying "this should work now." I'm confident that readme file also needs a once over because so many things are just unclear, such as the download kernel source line, I'm not told where to place both the msm-kernel and the kernel-common, and left to assume something that's pretty important to building. 

I also mentioned they have us CD into the common folder without actually telling us how that folder even exists, because it's never made with a mkdir, and instead we're supposed to know what goes there. That seems pretty bad considering the readme file is supposed to help users who may not be as knowledgeable as someone like you, and need that extra clarification.

This process of having us tell them issues is simply just bad support, as if you're giving building instructions, I'd check if they actually work and get you a successful build, instead of waiting for us to tell them that "it didn't work" and then only fixing a single issue or only the ones that were mentioned. The spelling errors that were made originally wouldn't have happened if the file was properly proofread before it was posted,  and also the errors that we encountered originally also wouldn't have happened if they followed the instructions and checked if they produced a build or not.

I'll be truthful and say that I'm definitely more lenient on the bad service, as I do understand they are busy doing other things in production/etc, but eventually I will just have to bring it up with the SFC if this doesn't improve because as I said before, this is very annoying for someone who's been trying this for the past 8 months/or more. Please, do better Motorola. This shouldn't have to be a problem with an open source company to begin with, and in fact makes me less likely to get another Motorola device when I eventually upgrade to a different phone due to this experience with them which I hate to say because I do enjoy this device and I want to continue staying with Motorola as they've made the process for unlocking the bootloader quite simple.

But also the fact that they call themselves open source when you have to end up making a request to even get a kernel, and when you do, it ends up taking weeks or a month at max before the source is uploaded, and by then there's already an update for a newer one that you need a kernel for again.

This isn't acceptable, and I wish that would change because this is how you lose potential customers, and also give your brand a bad reputation to certain buyers. This should be known, and should be important to them but yet it seems it isn't considering the support being given. 

Improvement is needed, and that shouldn't have to be said for an "open source" company. Just, please do better.

Lost-Entrepreneur439, 2025-07-18:
> [@lunarriaa](https://github.com/lunarriaa) Please create a symlink and make sure the tool points to the correct path. If you don't know how to do it, refer to the updated documentation. Updated the readme as you need: https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-U2UANS34.72-43-1-8.txt Thanks

```
ivy@debian:~/kernel_platform/external/dtc$ make all install CC=clang AR=llvm-ar LDFLAGS="--sysroot=$my_top_dir/build/build-tools/sysroot -Wl,-rpath,$my_top_dir/prebuilts/kernel-build-tools/linux-x86/lib64 -L $my_top_dir/prebuilts/kernel-build-tools/linux-x86/lib64 -fuse-ld=lld --rtlib=compiler-rt -fuse-ld=lld --rtlib=compiler-rt" EXTRA_CFLAGS="--sysroot=$my_top_dir/build/build-tools/sysroot -I$my_top_dir/prebuilts/kernel-build-tools/linux-x86/include" PKG_CONFIG=false NO_PYTHON=1 PREFIX=$my_top_dir/out/msm-waipio-parrot-gki/host
         DEP checks.c
         DEP data.c
         DEP dtc.c
         DEP flattree.c
         DEP fstree.c
         DEP livetree.c
         DEP srcpos.c
         DEP treesource.c
         DEP util.c
         LEX dtc-lexer.lex.c
         DEP dtc-lexer.lex.c
         BISON dtc-parser.tab.c
/home/ivy/kernel_platform/build/build-tools/path/linux-x86/bison: 7: /home/ivy/kernel_platform/build/build-tools/path/linux-x86/../../../../prebuilts/build-tools/linux-x86/bin/bison: not found
         LEX convert-dtsv0-lexer.lex.c
         DEP convert-dtsv0-lexer.lex.c
         DEP fdtdump.c
         DEP fdtget.c
         DEP fdtput.c
         DEP fdtoverlay.c
         DEP tests/get_mem_rsv.c
         DEP tests/root_node.c
         DEP tests/find_property.c
         DEP tests/subnode_offset.c
         DEP tests/path_offset.c
         DEP tests/get_name.c
         DEP tests/getprop.c
         DEP tests/get_prop_offset.c
         DEP tests/get_phandle.c
         DEP tests/get_path.c
         DEP tests/supernode_atdepth_offset.c
         DEP tests/parent_offset.c
         DEP tests/node_offset_by_prop_value.c
         DEP tests/node_offset_by_phandle.c
         DEP tests/node_check_compatible.c
         DEP tests/node_offset_by_compatible.c
         DEP tests/get_alias.c
         DEP tests/char_literal.c
         DEP tests/sized_cells.c
         DEP tests/notfound.c
         DEP tests/addr_size_cells.c
         DEP tests/addr_size_cells2.c
         DEP tests/appendprop_addrrange.c
         DEP tests/stringlist.c
         DEP tests/setprop_inplace.c
         DEP tests/nop_property.c
         DEP tests/nop_node.c
         DEP tests/sw_tree1.c
         DEP tests/sw_states.c
         DEP tests/move_and_save.c
         DEP tests/mangle-layout.c
         DEP tests/nopulate.c
         DEP tests/open_pack.c
         DEP tests/rw_tree1.c
         DEP tests/rw_oom.c
         DEP tests/set_name.c
         DEP tests/setprop.c
         DEP tests/del_property.c
         DEP tests/del_node.c
         DEP tests/appendprop1.c
         DEP tests/appendprop2.c
         DEP tests/propname_escapes.c
         DEP tests/string_escapes.c
         DEP tests/references.c
         DEP tests/path-references.c
         DEP tests/phandle_format.c
         DEP tests/boot-cpuid.c
         DEP tests/incbin.c
         DEP tests/extra-terminating-null.c
         DEP tests/dtbs_equal_ordered.c
         DEP tests/dtb_reverse.c
         DEP tests/dtbs_equal_unordered.c
         DEP tests/add_subnode_with_nops.c
         DEP tests/path_offset_aliases.c
         DEP tests/utilfdt_test.c
         DEP tests/integer-expressions.c
         DEP tests/property_iterate.c
         DEP tests/subnode_iterate.c
         DEP tests/overlay.c
         DEP tests/overlay_bad_fixup.c
         DEP tests/check_path.c
         DEP tests/check_header.c
         DEP tests/check_full.c
         DEP tests/fs_tree1.c
         DEP tests/truncated_property.c
         DEP tests/truncated_string.c
         DEP tests/truncated_memrsv.c
         DEP tests/asm_tree_dump.c
         DEP tests/value-labels.c
         DEP tests/testutils.c
         DEP tests/trees.S
         DEP tests/dumptrees.c
         BISON dtc-parser.tab.h
/home/ivy/kernel_platform/build/build-tools/path/linux-x86/bison: 7: /home/ivy/kernel_platform/build/build-tools/path/linux-x86/../../../../prebuilts/build-tools/linux-x86/bin/bison: not found
         CC srcpos.o
         CC util.o
         CC convert-dtsv0-lexer.lex.o
         LD convert-dtsv0
         CC dtc.o
         CC checks.o
         CC data.o
         CC flattree.o
         CC fstree.o
         CC livetree.o
         CC treesource.o
make: *** No rule to make target 'dtc-parser.tab.h', needed by 'dtc-lexer.lex.o'.  Stop.
ivy@debian:~/kernel_platform/external/dtc$ 
```
It's still broken.

lunarriaa, 2025-07-19:
> > [@lunarriaa](https://github.com/lunarriaa) ， I have checked the link files under the $my_top_dir/build/build-tools/path/linux-x86 directory, and they are indeed incomplete. I checked the source code and found that it originated from another Git repository, but the development team had merged it internally. I updated some commands, and the soft link needs to be implemented through scripts.
> > 
> > Command as bellow： """ cd $my_top_dir/prebuilts git clone -b ks-kernel.lnx.1.0.r1-rel https://git.codelinaro.org/clo/la/kernel_platform/prebuilts/build-tools.git SOURCE_DIRS=( "$my_top_dir/prebuilts/build-tools/linux-x86/bin" "$my_top_dir/prebuilts/build-tools/path/linux-x86" )
> > 
> > for current_source_dir in "${SOURCE_DIRS[@]}"; do if [ -d "${current_source_dir}" ]; then for file_path in "${current_source_dir}"/*; do if [ -f "$file_path" ]; then filename=$(basename "$file_path") ln -sf "../${current_source_dir}/${filename}" "$my_top_dir/build/build-tools/path/linux-x86/${filename}" fi done fi done """ Hopefully this will fix the problem.
> 
> well, it finally builds, but it just dumps out an image.gz, you don't get a boot.img, so it can't be flashed on the phone. can you provide the instructions to generate the boot.img too so the kernel can actually be flashed onto the phone? Without that, you're still violating the GPL.

Looked it up, this one's not on Motorola, in fact it seems common for kernels to compile into a gz, (as seen at least according to [this XDA post](https://xdaforums.com/t/tutorial-kernel-post-compilation-guide-how-to-turn-kernel-into-img.3604083/))

I believe for the first steps they want you to download the original boot.img from https://mirrors.lolinet.com/firmware/lenomola/2024/avatrn/ and unpack it using Osm0sis's mkbootimg, which can be found at https://github.com/osm0sis/mkbootimg Be warned, you do have to build it, and it may require troubleshooting, but it's not that bad and it really only might take a hour or two. For me, the troubleshooting was due to not having the correct dependencies, but I'd assume by now you probably have the ones needed.

I will specify here that you should not use any other version of mkbootimg, and only use Osm0sis's, as I've learned the hard way that some of them don't have the latest updates that actually support the header version.

Also unfortunately I can't build the kernel myself as the laptop I'm using does not have enough RAM/processing power to get past line 89 because it keeps getting killed by Linux when building vmlinux.o, but hopefully this works for you, but I can build mkbootimg if you do have issues.

Lost-Entrepreneur439, 2025-07-21:
@chenyt9 @lunarriaa 

i did try making the bootimg with that mkbootimg, but no, when i tried to flash it, phone just hung on the powered by android screen forever, never got to bootanim or connected to my computer.

im unsure if mkbootimg is failing or if motorola's kernel source is missing/broken (knowing motorola, probably the latter), but until motorola themselves provides instructions on how to get a boot.img, there's no real way to know.

lunarriaa, 2025-07-22:
> @chenyt9 @lunarriaa
>
> i did try making the bootimg with that mkbootimg, but no, when i tried to flash it, phone just hung on the powered by android screen forever, never got to bootanim or connected [...]

If I remember correctly I may have actually did mkbootimg on the vendor_boot which had more parameters and actually let me get a dtc, compared to boot.img which didn't really have much output.

Another thing I'll ask is if you disabled the vbmeta verity, as something similar happened when I would flash ROMs. You just do fastboot flash vbmeta vbmeta.img --disable-verity --disable-verification and reboot back into the bootloader or recovery and clear the dalvik cache, or it you know more than me, can clear that via fastboot and that may solve the problem.

Can you upload the gz so I can tinker with it?
Might be able to get something, but if I end up having the same result my guess would be something in the kernel being wrong, (most likely that bison warning when making dtc) or a step failed or something.

When you did line 89 did you have any errors? Because I did, mainly being dependency related but I ask that because for me it took a while to finish and I know you probably weren't looking at the terminal the entire time it was building because I know I wouldn't and what I found interesting about when I was building is that it would error sometimes but still keep going, which is not bad on a slow PC but with a fast one I'd assume you'd barely have time to notice the message before it washed away from the log

Also considering that we've gotten way closer to getting this done I'd suggest you check your Reddit DMs because I sent you a message a while ago and I think we should probably only post here if we're confident it's an issue on Motorola's side. (If you don't wanna use Reddit though I do have my Discord as one of my URLs on my GitHub profile, you should just have to click the link and my profile should just appear as if you clicked on my full profile)